Ebiroll/esp32_sigrok #1: Building sigrok issue due libsigrok dependency

EUA, 2018-10-28:
Hi.
2 days ago I tried to use your program, unfortunately without success... Whan I read that 10Ksps limit and just break to try to work on it...

Even I already build mine own version of ESP32 Logic Analyzer for higher sampling rates,
I want to say hello to you and make you aware from this issue.

with using **git clone git://sigrok.org/libsigrok.git** , compilation give me this error.
```
...
-- Configuring done
CMake Error at CMakeLists.txt:165 (add_executable):
  Cannot find source file:

    libsigrok/src/scpi/helpers.c

  Tried extensions .c .C .c++ .cc .cpp .cxx .cu .m .M .mm .h .hh .h++ .hm
  .hpp .hxx .in .txx

CMake Error at CMakeLists.txt:165 (add_executable):
  No SOURCES given to target: olas-cli

-- Build files have been written to: /home/compile/esp32_sigrok/sigrok/build

```

Ebiroll, 2018-12-02:
Oh regarding your Logic analyzer. Did you publish the source? Did you see this. https://www.labfruits.com/esp32-adc-speed/ ?

EUA, 2018-12-14:
Hello.
Sorry for lateness. Just busy with reversing some AC unit remote controller protocol for IRremoteESP8266 project this days... Due that i just paused to development of my ESP32 Logic Analyzer. I don't release any code right now. Since my IRremoteESP8266 work nears end, I probably will release codes in this month.

Well, I don't go on ADC stuff, yet. But done Digital Logic Analyzer only.
Using Arduino IDE for easy compile and load. I done and successfully sample up to 10Mhz without an issue using inline assembly and with counting clocks :D . Now figured out to sample 20Mhz also using **loopnez** ASM command. That save me to check loop constants. Unfortunately Interrupts are not working proper for first or first two samples in 20Mhz. 

For now I researching and trying to implement DMA which is better approach but looks little confusing :)

EUA, 2020-02-03:
Hi!
Late is better never.
At least I find time to complete my progress.
My implementation is only for digital input.
I leave timing based approach and use DMA for the job,
Bests,

https://github.com/EUA/ESP32_LogicAnalyzer

Ebiroll, 2020-02-03:
Nice job. Do you generate the pixel clock for the sample points?
Trying to understand the code.

EUA, 2020-02-03:
Yes, I do.
I use ledc to generate sampling frequency at PIN 23 and feed that signal via pin physically to PIN22 as a sig_clk.

I believe it there has to be a way to do it internally, without using physical pin for signal feeding.
If I found it, will add.

I also make another code (as I previously mentioned), using assembly to count/delay exact clock of the CPU but I think I have some issues like jitter on 20Mbps sampling with it...

I wish sigrok add OLS over TCP soon. (https://sigrok.org/bugzilla/show_bug.cgi?id=1490)