Ebiroll/esp32_sigrok #4: Clarify how to set up for basic digital use?

lf-, 2019-05-16:
I read the README and it indicates that I need to mess around with some (outdated) code lines in `main/` somewhere in order to get digital capture. Well, I did that, flashed and can't get sigrok-cli to get any response from the ESP32 using the command line from the README, which makes me feel like I modified these lines incorrectly. Additionally, the README talks about different serial lines but it isn't clear without substantial context that they are test data.

If there were a clear config system which made it possible to do anything "simple", that would be fantastic.

Ebiroll, 2019-05-16:
I am glad you asked. I also agree with you and will test clarify and make the different options into separate defines. The results have also been varying between different versions of esp-idf. The last thing I did was output on the display. I think that is what is stopping you right now.

Ebiroll, 2019-05-16:
You should end up with this
    //scpi_server_init(&xHandlingTask);
    sump_init();
    //sump_server_init();
    sump_uart();

Then you can test with this,

Ebiroll, 2019-05-18:
I tested, and it works but note that if you are using the WROVER board, you must use USB1 as USB0 is used by the FTDI chip:

./sigrok/build/olas-cli -d ols:conn=/dev/ttyUSB1 -l 5 --scan

Ebiroll, 2019-05-18:
I added the defines and tested with the UART output. Changes are pushed to the repository. The next step is to add i2s sampling as 10kHz is not so good.

closed 2019-09-12 by Ebiroll