Ebiroll/esp32_sigrok #5: driver uart error if enable SUMP_OVER_UART

almoondsllc, 2019-10-05:
I would like to use SUMP over uart but I got this error when I enable it on app-config.h

Ebiroll, 2019-10-07:
Hello, I cannot see the error. But I would agree that it might not work. 
You might asl have to disable SPCI.
//#define  SCPI_ON_NETWORK 1
When I run in qemu I get,
E (216) uart: uart_flush_input(1281): uart driver error
E (216) uart: uart_read_bytes(1213): uart driver error
But this has been fixed with a very recent commit.
I have only tested with qemu, but will test when I have time. It might work.

Ebiroll, 2019-10-08:
I tested it on a Wroover board and it works for me with SUMP over UART. Make sure you set sampling speed to 10kHz when using pulseview.

closed 2019-10-09 by Ebiroll