Ebiroll/esp32_sigrok #6: Compatibility with esp32-idf 4.0+

Civil, 2020-04-19:
esp32-idf-4.0 removed some of deprecated functions:

 * `esp_adc_cal_get_characteristics` was removed in favor to `esp_adc_cal_characterize` (parameters are in different order, but otherwise the same)
 * `i2s_write_bytes` was removed (could be safely copypasted from pre-4.0 sources as it's basically wrapper around `i2s_write` with extra checking)
 * Field `bit_num` in `struct ledc_timer_config_t`  was renamed to `duty_resolution`
 * `components/tft/tftspi.c` must explicitly include `drivers/gpio.h`

I'm not sure that after those fixes it works properly though...

Ebiroll, 2020-04-29:
Hello, I commited a change that allows to compile with latest version of esp-idf. It should probably work for esp32-idf-4.0 as well

bessobesso, 2020-05-02:
Hi, tried to compile with idf 4.0 and it throw errors.  Did you test to compile with idf 4, with idf.py build?

Civil, 2020-05-05:
@Ebiroll it doesn't compile, fails with:
```
/home/civil/CLionProjects/esp32/esp32_sigrok/main/analog-testdata.c: In function 'example_i2s_adc_dac':
/home/civil/CLionProjects/esp32/esp32_sigrok/main/analog-testdata.c:250:13: error: implicit declaration of function 'i2s_write_bytes'; did you mean 'i2s_write_expand'? [-Werror=implicit-function-declaration]
             i2s_write_bytes(EXAMPLE_I2S_NUM, (const char*) i2s_write_buff, i2s_wr_len, portMAX_DELAY);
             ^~~~~~~~~~~~~~~
             i2s_write_expand

```

If you fix that, it would fail with:
```
/home/civil/CLionProjects/esp32/esp32_sigrok/main/espScope.cpp:97:16: error: 'struct ledc_timer_config_t' has no member named 'bit_num'; did you mean 'timer_num'?
     timer_conf.bit_num = (ledc_timer_bit_t)1;
                ^~~~~~~
                timer_num
```

Civil, 2020-05-05:
I'd make a PR with the changes I've made, but I technically cannot as your project doesn't have a license, so technically is not open-source.

Ebiroll, 2020-05-05:
OK. I have commited some changes. Hopefully it will compile now.
Technically I do not know what license to use, as I have code with different licenses.
The way I try to  move the project forward is to use the i2s camera mode with the pixel clock (samling times) generated by the ledc pwm. Data should then be sent over the network to a proxy process that talks to sigrok. Any other ideas?

bessobesso, 2020-05-06:
I am able to compile now, but it does not appear in PulsView, see attached screenshot of my PulsView setup window

[image]

Ebiroll, 2020-05-06:
You must enable sump in main/app-config.h
Also baudrate should be 115200.
It helps to watch boot messages in a terminal before using pulseview.
Also take a look at this project https://github.com/EUA/ESP32_LogicAnalyzer

Civil, 2020-05-06:
@Ebiroll about the License - I'm not a lawyer, but here's my opinion:

**TLDR;** But overall I would suggest to choose between BSD, MIT, Apache2, GPLv2, GPLv3, LGPLv2 and LGPLv3. If you've used code that's distributed under any GPL license you basically have no choice other than to adopt GPL for the whole project, but otherwise you can mostly freely choose.

Rather good comparison/wizard is: https://choosealicense.com/ (github's resource to my knowledge).

Otherwise your code falls into https://choosealicense.com/no-permission/#for-users which is tricky for people to use/contribute.

**Long version:**
You should take something that is compatible with all other code that you reuse.

Good examples are GPLv2/v3, LGPLv2/v3, Apache2, etc. It also depends on what you want.

Different licenses have different requirements, for example famously GPL overrides all other licenses to become GPL (you still can use libraries or parts of the code licensed under BSD/MIT and other compatible licenses). BSD/MIT/Apache are more permissive (basically they allows other people to use your work as part of their closed source project, with small restrictions, like they need to put that statement somewhere that this is the code they use and depends on version it also covers the name of your application and other stuff).

There is another class of licenses which are tricky to actually use - those are WTFPL-like licenses. Their problem is that depends on a country they might be considered not a valid license (WTFPL even the updated version is not considered a valid license in US and some other countries, some other countries could not recognise CC0 license, etc).

And the last one is even more strict than GPL, which is AGPLv3. With GPL there is a "problem" that you can use GPL code in a backend of your application and sell your work as SaaS without providing source code of your backend. AGPLv3 fixes that problem. The problem is that for such licenses some external people have restrictions in what they can contribute to. E.x. I cannot contribute to projects under AGPLv3 even in my free time without getting approval for specific project. And that's true not only for me, but that's rather common restriction imposed by US-based IT Companies.

Oh, and another thing, most licenses limits liability and warranty, which is something that you might want as well (to not be responsible for your application's behavior or damages caused by it).

bessobesso, 2020-05-06:
I tried now with 115200, and sump(SUMP_OVER_UART) is enabled/defined  in app-config.h
Stil cant connect to PulsView

Bootlog looks like this

I (106) cpu_start: Pro cpu up.
I (106) cpu_start: Application information:
I (106) cpu_start: Project name:     sigrok
I (108) cpu_start: App version:      1
I (113) cpu_start: Compile time:     May  6 2020 18:15:15
I (119) cpu_start: ELF file SHA256:  15cb9cdad376a204...
I (125) cpu_start: ESP-IDF:          v4.0-dirty
I (130) cpu_start: Starting app cpu, entry point is 0x40081150
0x40081150: call_start_cpu1 at C:/Users/besik/SDKs/esp-idf/components/esp32/cpu_start.c:272

I (0) cpu_start: App cpu up.
I (141) heap_init: Initializing. RAM available for dynamic allocation:
I (147) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM
I (153) heap_init: At 3FFB3310 len 0002CCF0 (179 KiB): DRAM
I (160) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (166) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (172) heap_init: At 400899B4 len 0001664C (89 KiB): IRAM
I (179) cpu_start: Pro cpu start user code
I (197) spi_flash: detected chip: generic
I (198) spi_flash: flash io: dio
W (198) spi_flash: Detected size(4096k) larger than the size in the binary image header(2048k). Using the size in the binary image header.
I (208) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
I (283) sigrok: Setting UART configuration number 0...
I (293) uart: queue free spaces: 20
I (293) sigrok: Setting UART configuration number 1...
I (293) uart: queue free spaces: 10
sump UUUI (293) gpio: GPIO[12]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0
I (303) gpio: GPIO[13]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0
I (313) gpio: GPIO[14]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0
I (323) gpio: GPIO[15]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0