Ebiroll/esp32_sigrok #7: binary

valoni, 2020-06-03:
can you publish compiled binary file for ESP32-DevKit1 or NodeMCU32 ...
(and some info what file / what channel to be used) ...

regards,
VALONI

Ebiroll, 2020-06-04:
Yes. That is a good Idea. The quality of the binary also depends very much on the version of esp-idf. The problem has been wifi passwords, hat needs to go into the build. Espressif has some smartphone tool to aid the process, https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/provisioning/provisioning.html but I think it would be better if you could add password over a serial terminal. What is your opinion?

Ebiroll, 2020-06-04:
Also, I can recommend this. https://github.com/EUA/ESP32_LogicAnalyzer Make sure you pull GPIO15 to ground.

valoni, 2020-06-04:
> Also, I can recommend this. https://github.com/EUA/ESP32_LogicAnalyzer Make sure you pull GPIO15 to ground.

tried to compile it but i've got error during compile can you please can compile both files for me and provide ready bin files to install at 0x00

PR #8 (2020-06-09): Added CI Build 
Added CI build using Github actions. 
Artifacts (firmware.bin) are available on the actions page. Closes #7 
Added Badge (Not sure if you'll need to update first page with your link or it will be done automatically).
closed 2020-06-09 by Ebiroll