Ebiroll/esp32_sigrok #10: Provisioning of wifi to allow prebuilt binary

Ebiroll, 2020-06-09:
Try to get a simplified input of SSID and password to allow a prebuilt binary

https://github.com/espressif/esp-idf/blob/master/examples/provisioning/legacy/console_prov

saikek, 2020-06-11:
Can you try this: https://docs.platformio.org/en/latest/projectconf/section_env_build.html

```
[env:string_defines]
build_flags =
  -DHELLO="World!"
  '-DWIFI_PASS="My password"'
  ; Password with special chars: My pass'word
  -DWIFI_PASS=\"My\ pass\'word\"
```

And probably in code 
```
#ifndef variable then assign default value
```

saikek, 2020-06-11:
Other option is to have USER_CONFIG file that users can update. Anyone who forks repo and edits this file in github would have firmware artifact built by actions.