Ebiroll/esp32_sigrok #12: Use with M5Stack Atom

tbaumann, 2022-01-05:
Can someone help me to figure out how to use this on a M5Stack Atom?

I set those values in `app-config.h`
```C
#undef UART_TEST_OUTPUT
#define SUMP_OVER_UART 1
#undef SCPI_ON_NETWORK
#undef SUMP_ON_NETWORK
#define UART_OUTPUT_PIN 1
#define UART_RX_PIN 3
#define PARALLEL_0  38 // Selected for a range around pin 39
```
It took a while but `sigrok-cli -d ols:conn=/dev/ttyUSB0 -l 5 --scan` discovers the device now mostly reliably.

I would have expected to get a trace for my push-button connected to GPIO39. But I see nothing when capturing with pulseview. 

Here the official pinout
[image]

My ultimate goal is to analyse the timing of an IR receiver on pin GPIO25. But I thought I'd start with something simple like the push-button at 10Hz sampling rate.

PS: The readme should perhaps mention that the project can just be built with plattform.io

PPS: Do you suggest I try to understand how to use SUMP via network instead?

Ebiroll, 2022-01-05:
Hello, I think pins above 32 might cause problem as data is read with a "hack" from a regsister

tbaumann, 2022-01-05:
I thought maybe the naming and numbering is different. But I don't think so. Not sure how to check.
Even the esp32-pico manual from Espressive lists only the high port numbers in the manual.

Pitty. I have just recently moved and don't have access to any tools. I don't even have a soldering iron. :laughing: 
I thought I could McGyver myself out of this without even any tools.

tbaumann, 2022-01-05:
Actually, I'm interested in GPIO25. That should be in the good range. Testing with `#define PARALLEL_0  23`...

tbaumann, 2022-01-05:
Nothing, except always a bit pattern in the few frames.

[image]

Ebiroll, 2022-01-05:
Oh. Looks familiar. Sorry, I cant help more there. Also look at the links for other ESP32 SUMP projects, maybe they work better.

Ebiroll, 2022-01-05:
https://github.com/EUA/ESP32_LogicAnalyzer

tbaumann, 2022-01-05:
> https://github.com/EUA/ESP32_LogicAnalyzer

I struggled getting that one to even connect. :D But thanks for the hint.

closed 2022-01-05 by tbaumann